bambulab/BambuStudio #6542: Can't send job to printer

igor-gia, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel Core i-5 8265U
RAM: 16Gb
SSD: 1Tb
Video: MX-130

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Everything worked fine
2. An update has arrived to 2.0.2.57
3. After installing this update a problem occurred - after models slicing , I press - send to print. The printer is visible, but the button is not active

### Actual results

the program became inoperative
I launched the orca slicer - everything works from it

### Expected results

the program became inoperative
I launched the orca slicer - everything works from it

### Project file & Debug log uploads

It is impossible to send any project to printer

**your update is broken**

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2025-04-16:
Thanks for your feedback and can you upload your log for us to check ?https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

igor-gia, 2025-04-16:
[log.zip](https://github.com/user-attachments/files/19778925/log.zip)

Thank you for your feedback.
I attach zip-archive with log files.

Looks like I found the cause of the problem.

I updated to version 2.0.2.57.
I slice the model, click “Prepare Build Plate,” but the “Send Plate G-code to SD Card” button is inactive.
I launched OrcaSlicer — everything prints just fine from there.
I restarted the printer and PC, reinstalled Bambu Studio — no luck.
I spent a long time looking for the reason, but I finally found it.
I usually changed filament like this: Unload, then Load. When slicing, I always selected the correct filament type. Everything worked fine until today’s Bambu Studio update.
Now, **if the filament type set on the printer doesn't match the one used during slicing in Bambu Studio, it won’t let you print.**
Having to tap through the small screen every time to set the filament is inconvenient. I couldn’t find an option in Bambu Studio to disable this check.

XunZhangBambu, 2025-04-21:
@igor-gia I think this is the correct behavior. Allowing mapping to different filament  types would not be a reasonable approach for most users. You can sync filaments from AMS before slicing.

[image]

igor-gia, 2025-04-21:
Thank you for your reply.
This behavior is OK when AMS is installed. But without AMS, I already know which filament I use — so what is there to synchronize?
Also, I have to set the filament type two times: once in the printer settings, and again in the slicer. This looks unnecessary.