bambulab/BambuStudio #6544: Latest Bambu Studio v02.00.02.57 crashes

xtiangriffin, 2025-04-16:
### Bambu Studio Version

v02.00.02.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home v22H2

### Additional system information

Enhetsnamn:	ACER XC-603
Processor:	Intel(R) Pentium(R) CPU  J2900  @ 2.41GHz   2.41 GHz
Installerat RAM-minne:	8,00 GB (7,88 GB går att använda)
Enhets-ID:	EF05BD25-386B-4522-8CA1-49B011E66F1D
Produkt-ID:	00326-10000-00000-AA003
Systemtyp:	64-bitars operativsystem, x64-baserad processor

### Printer

Bambu Lab X1C

### How to reproduce

When I exit Bambu Studio It crashes and I get a message, "Unhandled unknown exception; terminating the application."
This crash only occurs with v02.00.02.57, not with v02.00.01.50.

### Actual results

[image]

### Expected results

I must be able to exit Bambu Studio without crashing.

### Project file & Debug log uploads

[crash_Wed_Apr_16_14_44_32_0_done.log](https://github.com/user-attachments/files/19777451/crash_Wed_Apr_16_14_44_32_0_done.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

MartinChan3, 2025-04-16:
Thanks for your feedback,and can you offer 3mf or MakerWorld link for us to give it a test?

xtiangriffin, 2025-04-16:
Yes, of course...
[AMS-Drybox PROJ.zip](https://github.com/user-attachments/files/19778804/AMS-Drybox.PROJ.zip)

Haidiye00, 2025-04-17:
[image]
Hello, can you package all the files in this log folder, or the latest log and dmp files. We were unable to reproduce your issue.

xtiangriffin, 2025-04-17:
[dump-log.zip](https://github.com/user-attachments/files/19787284/dump-log.zip)

I can add that the crash happens regardless of which previous project I have opened.
If I create a new project, and load stl files, it works.

JohnSmithCYM, 2025-04-17:
I have tried this project both on mac and windows and it also slice okay.Do you do some opeartion before that? And which language are you using?

Haidiye00, 2025-04-17:
Hello,Can you take some more time to help us record your rough operation? Thank you. Start recording by opening the software to import the project.
 can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

xtiangriffin, 2025-04-17:
[image]

xtiangriffin, 2025-04-17:
I just discovered that if I have opened a project, then save it.. it works.
.. but if I then open another project while I'm inside Bambu Studio, it crashes immediately.

xtiangriffin, 2025-04-17:
Hi,
My Bambu Lab X1-Carbon is connected to wifi.
firmware, X1C: 01.08.05.00 - AMS: 00.00.06.44

I logged out of the network and my account and logged back in..
..but no difference. same issue.

xtiangriffin, 2025-04-17:
Isn't there a configuration file that I can delete, which Bambu Studio then recreate?

I'm wondering because I find it takes longer to start Bambu Studio now than with the previous version.

xtiangriffin, 2025-04-17:
Yes, I still have the same crash without the machine connected.

xtiangriffin, 2025-04-17:
Hi again,
I've tested a little more..
I don't know if this can be of any help...
(a long time ago I worked quite a lot as a Beta tester for machines and their software).

so,
It's something with 3mf files that are created in previous versions of Bambu Studio.

If I create a new project, load some stl files and then save this project that was created in v02.00.02.57, it works as it should!

..but as soon as I open a project that was created in v02.00.01.50 or earlier, Bambu Studio crashes.

also, If I load one of these previous projects, change something then save, that works, but if I then choose to open another project, the program show the message and crashes when I click OK.

JohnSmithCYM, 2025-04-20:
Thanks for your feedback. Can you still offer your dmp file and log file for us(? We will chceck if every time the crash happened at the same place.
And before we released the whole version, We checked 1.10 3mf import and no obvious crash happened so your problem is so wired.
Can you upload your all config file(like what in the red frame below). You can zip it and send to [malkey.chen@bambulab.com](mailto:malkey.chen@bambulab.com), we make sure no data leak and the only use is to find out this annoying problem.
[image]

xtiangriffin, 2025-04-20:
Thanks for all your help.

I send you the files directly.

/Christian

xtiangriffin, 2025-04-22:
Hi again,
This is very very strange.
I have a brand new laptop (4 months old) so I installed the latest version of Bambu Studio on it.... and the same result!??
I get the error message.
But as previously said, Only if I open an old project.

JohnSmithCYM, 2025-04-22:
We have detected your crash point,and we will solve it in the next version.Wait for that and give it another try.

xtiangriffin, 2025-04-22:
Thank you very much!

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

closed 2025-04-25 by Haidiye00

xtiangriffin, 2025-04-25:
Hi,
Now it works flawlessly!

Thank you very very much, now it is much easier to work with Bambu Studio.

I must say that you really have a very good support and are very professional!
Thank you.

I also have a question,
I am a 64 year old guy, who lives in Stockholm Sweden, is a photographer and has worked with 3D graphics for 40 years and has now used my X1C for a couple of years.

I have looked a little at the new H2D with double heads and AMS.

I already have a laser and cutter so it is not relevant.

Now to my question;
Is it possible to test and review an H2D here in Sweden?
I don't know if you have anyone who tests in Sweden here in the Nordics?

Thanks again.

Best regards,
Christian Grip

Haidiye00, 2025-04-25:
Hello, thank you very much for your recognition. Our sales team will coordinate with you. Thank you

xtiangriffin, 2025-04-30:
Hi,

I want to start by saying that the latest version of Bambu Studio works very well!
The latest version for X1C also works well and I really like the new graphical interface.

Now to a small error that has crept into Bambu Studio..
In the Swedish translation in Bambu Studio when you open a project, a letter is missing.

Project is called the same in Swedish but is spelled "Projekt".

When you select "New Project" from the menu, it says "Pojekt",
and when you select "Öppna Project" it is spelled correctly "Open Projekt".

So there is an "r" missing.

That was all.
Have a nice weekend!