bambulab/BambuStudio #6545: Updated Bambu Studio and now can't print anything!

Gerryrys, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Once a project is sliced I go to print and I get a filament error regardless of what I try to print and with whatever fillament.  It was all working fine before the latest software update.

### Actual results

Can't print

[image]

### Expected results

To be able to print something!

### Project file & Debug log uploads

[BMW K25 Frame Big.zip](https://github.com/user-attachments/files/19778978/BMW.K25.Frame.Big.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tongzhou-j, 2025-04-17:
我也遇到了

MartinChan3, 2025-04-17:
Can you just click the green PLA filament and see if you can map to your filament.You should make sure the mapped filament should be the same type with slice preset.

Haidiye00, 2025-04-17:
[image]
[image]
Hello, the consumables type on your device page must be the same as the consumables type on the page you are preparing in order to initiate printing.

Gerryrys, 2025-04-17:
THanks, it works now.  It seems you have added another element to print setup without telling anyone!  You now have to set the filament parameter in two different screens where previously, you only had to do it in one.

Haidiye00, 2025-04-17:
hello,May I ask if this situation occurs when printing fails: AMS is not installed, and the virtual slot is set to a filament type?

lobodpav, 2025-04-21:
Today, I literally spent an hour until I found this GitHub issue. Tried to print TPU, but the printer thinks I'm using PLA.

With the A1 mini, this makes no sense. I have an external spool - why do I have to change filaments in 2 places now?!

Fortunately, I found a workaround. It's still a pain in the ass, but better that choosing filaments on 2 places.
1. Go to `Device`
2. Click the small pen icon on the filament spool.
3. Click the `Reset` button. This will change your filament to `?`.
4. Now you can change your filament in the project to anything of your liking.
5. In the print dialogue, you can click the filament and select `Ext` (which was not possible previously).

[image]
[image]

walterwongbbl, 2025-04-24:
Thank you very much. If you have any other questions, please initiate a new conversation.

closed 2025-04-24 by walterwongbbl