bambulab/BambuStudio #6547: Comma not accepted as decimal point

oscarmarcelo, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sequoia

### Additional system information

Number input fields should always accept both dot (`.`) and comma (`,`) as a decimal points.
However, none of the number input fields in Bambu Studio accept the comma.

### Printer

A1

### How to reproduce

1. Import a model or create a primate;
2. Select one control that has number input fields (eg. Move, Rotate, Scale, ...);
3. Type (or paste) a decimal number  using comma as decimal point (eg. `4,5`).

### Actual results

The result is a number without a decimal point (eg. `45` instead of `4,5`, or even remapped to dot, like `4.5`).

### Expected results

Expected a decimal number, using either the original decimal point inserted by the user, or at least remapped to a dot (`.`). However, the best choice would be to follow the system preference instead of remapping to one of the options.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

biscuitlad, 2025-04-16:
A comma isn't a decimal point? What if someone writes 1,000,000?

oscarmarcelo, 2025-04-16:
@biscuitlad It should be handled anyway. I'm not familiar with the codebase, but I'm assuming that this part of the UI is using web technology, and if so, [Intl.NumberFormat](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Intl/NumberFormat) can parse it correctly, given the system locale.

So, any of these should be parsed correctly:
- `1,234,567.98` → `1234567.98`
- `1.234.567,98` → `1234567.98`

Haidiye00, 2025-04-18:
Hello, you can switch to languages such as German, which use commas as decimal points.