bambulab/BambuStudio #6549: Bamboo studio crashes when loading files over 1GB

biscuitlad, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Device name	Kester-workstation
Processor	Intel(R) Core(TM) i5-10600 CPU @ 3.30GHz   3.30 GHz
Installed RAM	32.0 GB
Device ID	DCF70FA6-7FE9-4AE2-ACD2-F49AEB4F6000
Product ID	00330-81490-83076-AA521
System type	64-bit operating system, x64-based processor
Pen and touch	Pen support
Edition	Windows 11 Pro
Version	24H2
Installed on	‎20/‎10/‎2024
OS build	26100.2605
Experience	Windows Feature Experience Pack 1000.26100.36.0

### Printer

A1

### How to reproduce

Add file.
Bambu Studio hangs during loading and has to be forcibly shut down by windows.

### Actual results

![Image](https://github.com/user-attachments/assets/b669cb16-2fe5-4

[image]

e33-a5ac-7bf2f4a4dde4)

### Expected results

It should add the stl file to the plate

### Project file & Debug log uploads

No project file as stl file fails to load. stl to large to include as an attachment.

Your system won't allow me to attach the log file? Says file type not allowed?

[debug_Wed_Apr_16_20_56_18_82188.log.zip](https://github.com/user-attachments/files/19784005/debug_Wed_Apr_16_20_56_18_82188.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-04-17:
Hello, can you provide this STL file? Thank you. Facilitating our analysis and resolution of problems.

biscuitlad, 2025-04-17:
I can't, it was 1.1GB in size. I've since reduced the size of the file and
it's ok, but you can reproduce by creating an object in Blender,
subdividing it many times, then duplicating the object as many times as you
need to create a file of that size.

Haidiye00, 2025-04-17:
Hello, are you saying that your STL file is as large as 1.1G? This is too big

Haidiye00, 2025-04-17:
I am curious about the number of meshes in Blender. If there are more than 5 million patches, our software will also lag. Suggest keeping STL files within 100M.

biscuitlad, 2025-04-17:
Yeah, I applied a surface displacement on a very dense mesh to several
objects and the face count went sky high!

All sorted now after I did a remesh.  ;-)

biscuitlad, 2025-04-17:
> Is there still a problem with importing now?

It's working now.

Haidiye00, 2025-04-17:
Okay, then I'll close the question. Thank you again for your feedback~

closed 2025-04-17 by Haidiye00