bambulab/BambuStudio #6550: Printing Cancelled Error message

GadgetGirl75, 2025-04-16:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 64 bit

### Printer

P1S

### How to reproduce

I cannot reproduce the message. Bambu Studio/Handy gave me an error that Printing was cancelled. The printer simply stopped in place mid print without returning to home and cannot be resumed. 

### Actual results

Print stops. Cannot view files on SD card. Cannot resume print as studio does not indicate that there IS a print in progress. 

### Expected results

Should be able to resume printing

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19785033/log.zip)

[SimpleTome_FlatBack_160_fixed socketsv1.2.3mf.zip](https://github.com/user-attachments/files/19785036/SimpleTome_FlatBack_160_fixed.socketsv1.2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-17:
Thank you for your feedback. Please update to the latest version V2.0.2.57 on our official website and give it a try. Download link: https://bambulab.cn/zh-cn/download/studio

GadgetGirl75, 2025-04-17:
Well given that it's taken 2 months for this problem to recur from the last time it happened, I doubt I'm going to see a difference. After a restart and calibration, It is printing normally this morning. I'm more wondering WHY it happened.