bambulab/BambuStudio #6551: Extruder keeps clogging after update

marcellatwin, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.3.2

### Printer

P1S

### How to reproduce

I haven't been able to reproduce it directly. Every time, but right now it is clogging on every filiment that is NOT from Bambu Lab.
I initially printed several small panel's to test out different versions of my companies logo.  The 2 colors were black ABS from Bambu and a white ABS from a large chain in my country (Clas Ohlson).  It printed fine enough, little rough but I hadn't done a flow calibration in a while.  After deciding on the final logo, I tried to print the final customer part which was 95% black ABS with just the logo in white ABS.  Every single time it "clogged" on the filament switch.
I ran out of time with the customer and had to print in the black ABS from Bambu and came out near perfect.  I didn't do anything to unclog, just canceled the print and restarted which it switched from the "clogged" white to the black no problem.
After the customer part was done, I went back to fix the issue.  I first cut the end off the white ABS so there's not blob.  I ran a calibration for flow and flow rate on the black ABS from Bambu just fine.  But the very first time I tried to do it with the white ABS it clogs every time.  I even tried to push it thru manually with the Handy app moving the extruder down and it went fine.  But when the AMS has to push it, even if I try to help it along or change the angle of the tube, it "stops" every time.
I tried the calibration on a PETG from Bambu and it went fine.  Then I tried it on a PLA from the Clas Ohlson brand and it clogged several times also, but was able to move it thru but then the calibration page froze on me like it has in the past and had to restart.
I've never had these problems before and I just switched to the Hardened nozzle recently so the nozzle isn't the problem.  But think all the issues started right after I updated the firmware for both the printer and Bambu Studio recently.

### Actual results

NA

### Expected results

NA

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-17:
From your description, the colleague advice you should find out whethter the filament or preset setting:
-  If you use ABS from other brand, gernic ABS preset is safer than Bambu ABS preset.
-  You can decreaset the temperature and speed  when printing ABS  in your situation.
May this could help you.

marcellatwin, 2025-04-17:
When I use the other brand, I DO use the Generic ABS settings.

The tiny print I did before the update was fine. But now the filament won’t
get past the extruder when it switches. It tried multiple times. I see the
extruder gear start to move it down then suddenly pushes the filament back
out and retracts all the way to the AMS. After a couple of tries it gives
the warning of possible clog.

But if I switch to the Bambi ABS it has no problem. I believe the only
change since that first print and now having issues is the FW update and
Studio update.

gaolegao-lx, 2025-04-21:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

marcellatwin, 2025-04-24:
Good morning,
I’m trying to find the logs but don’t see the same UI as you do. Attached
is what I see.
Also I have some updates. I tried a print with JUST Bambu filaments, dried
and basic PLA with something like 7 filament changes between the 2 colors
and every single change had a “clog”. After it tries 4-5 times with the
filament, it shows me the warning. Sometimes if I click the Problem solved,
resume printing button once or even twice it will be continue. Other times
I can get it to continue if I pause/unpause. And my last resort is somehow
clicking around during the pause to manually heat it up to a higher temp
and clicking the load button (I believe, trying several different things
while stressed I didn’t write down the exact sequence of button presses)
then it will work and I can press play to have it continue. I don’t have a
lot of options without canceling the print all together. Which isn’t an
option when some customer prints are 20-30 hours long.
Thanks again.
Jonathan Crossley

closed 2025-05-07 by BoBoYu-BamBu