bambulab/BambuStudio #6552: Bambu Lab Studio crash if a project is opened and you try to open another one

L0rdS474n, 2025-04-17:
### Bambu Studio Version

2.0.2 Public Release

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1, Open a project
2. Try to open another project (CTRL + O or from menu)
3. Bambu Lab Studio will crash with unhandled unknown exception

### Actual results

Studio is forced closed with an error message displayed

### Expected results

Being able to work in one project, save and open a new one and so on.

### Project file & Debug log uploads

Neither logfiles or model is needed since you can repeat this with any model on any system and for sure with any printer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-17:
Hello, can you provide these two 3mf?

L0rdS474n, 2025-04-19:
I can reproduce this without even opening a project.

1. Start a new project
2. Insert a 3DBenchy
3. Slice it
4. Press CTRL + N
5. Answer no for saving
6. Program crashed

This happens consistent with latest Studio, there is no specific 3mf file that crasch the studio but anything after a slice will kill the studio.

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

closed 2025-04-25 by Haidiye00