bambulab/BambuStudio #6555: BambuStudio is not scaled on the MAC Book Pro M2 (IOS), it is much too wide.

Vespisti1969, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.2 (24D81)

### Additional system information

Apple M2 Pro

### Printer

Bambu Lab X1C with two AMS

### How to reproduce

1. Start Bambu Studio
2. see the wrong scaling

[image]

### Actual results

nothing, wait for a solution

### Expected results

nothing, wait for a solution

### Project file & Debug log uploads

[debug_Wed_Apr_16_20_54_46_41019.log.0.zip](https://github.com/user-attachments/files/19788872/debug_Wed_Apr_16_20_54_46_41019.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-18:
Thank you for your feedback. Could you please tell me what the resolution of your computer is?

Vespisti1969, 2025-04-18:
Hi,The Resolution is 13“ Inch.If I start Bambu Studio, everything is well, but when I start the print, and I look to the printer site, there is the problem.Best regards