bambulab/BambuStudio #6558: You are trying to manage my MS Edge browser when I install bambu studio

xwzkj, 2025-04-17:
### Bambu Studio Version

2.0.2 and earlier

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

has nothing to do with this

### How to reproduce

just install

### Actual results

[image]
It adds a registry key that has nothing to do with itself, which will make my edge browser show that it is managed by the organization.

### Expected results

Don't do that!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

xwzkj, 2025-04-17:
Slicing software should only do its part.

lanewei120, 2025-04-17:
We’re very sorry for the trouble caused. Let me explain in detail why this change was made.

Bambu Studio uses Microsoft Edge WebView2 to load web content. Previously, some users reported issues where WebView2 failed to load. 
After investigation, we found the root cause was a conflict with certain third-party libraries (such as Green Shield). According to information we found online, modifying a specific registry key could resolve this issue. Importantly, this change does not affect the Microsoft Edge browser itself, as it uses the same registry key.

At the time, we rushed to fix the problem and didn’t fully consider all the implications.

In hindsight, we realize it would be more appropriate to let users decide whether to import this registry key when they encounter issues with WebView2 loading.

We will improve this in future versions. Thank you for your feedback.

AnalogKnight, 2025-04-18:
I agree, it should be optional because not every user will have problems. Even if Bambu Studio didn't do anything to my Edge, it would still be uncomfortable.

xwzkj, 2025-04-18:
That's true