bambulab/BambuStudio #6560: Print Speed in Bambu Studio is not shown correctly

JROJRO, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Start Bambu Studio
2. Send Job to printer
3. Start Print Job at printer
4. Set print speed manually to 50%

### Actual results

5. Get back to Bambu Studio
6. Select Device page
7. in Control overview 100% speed is shown, however when clicking on the Speed icon, 50% is shown and slider is located at the very left side (correctly).

[image]

### Expected results

 Just 100% speed indication at icon should be 50%.

### Project file & Debug log uploads

[BS-1xSet-Amini-Feeder_FilamentGuide_PETG_and_PC4-M10-One-Color_bug-report.zip](https://github.com/user-attachments/files/19792277/BS-1xSet-Amini-Feeder_FilamentGuide_PETG_and_PC4-M10-One-Color_bug-report.zip)

[debug_Thu_Apr_17_11_35_42_7368.log.zip](https://github.com/user-attachments/files/19792359/debug_Thu_Apr_17_11_35_42_7368.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-17:
@JROJRO Thank you for your feedback. As the printing speed is set during the machine's printing process, the machine needs a certain reaction time to make the corresponding setting operation, which is approximately 15 seconds. Please give it another try. If there is any problem, please let us know.

JROJRO, 2025-04-17:
Thank you for your quick reply. It seems not to be a temporal issue. There is no update of the Percentage in Bambu studio. Even not after 30 mins and more. Aditionally I have seen that also in the Android App the Percentage is still 100% even though the mode is correctly shown as "silent".