bambulab/BambuStudio #6561: Deletion of individual entries under "Filament Projects" when starting the application

FPrefect20, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Additional system information

Entries made under "Project Filemanete" will be reset when the application is restarted.

### Printer

Bambu Lab X1C

### How to reproduce

1. Section "prepare"
2. Set filament entries (alternatively: synchronize with AMS)

[image]

3. Restart Application

[image]

Note that the color is still saved, but the filament information is deleted (reset to the first entry in the list: "ABS")

### Actual results

Filament Information always deleted after re-run of the application.

### Expected results

Filament information stored

### Project file & Debug log uploads

on request 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-18:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

FPrefect20, 2025-04-18:
Sorry... I don't want to download / install software via links.
How can I explain it better?

It's really that simple:
- Synchronize with AMS
- Quit the application
- Restart the application
- Entries have disappeared from the application (pls. see Screenshots)

I have to resynchronize the 'filament information' after EVERY Bambu Studio startup. 
If I forget, it prints with the "ABS" settings! :-(

This behavior did not occur in the previous release.

FPrefect20, 2025-04-19:
Additional information:

When closing Bambu Studio after synchronizing with AMS, the config file is saved correctly:

[image]

However, after restarting and closing again, the information in the config file is overwritten:

[image]

FPrefect20, 2025-04-30:
Error still exists in Bambu Studio V. 2.0.3.54