bambulab/BambuStudio #6569: Cant select the external spool feed

RiverbassX, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

w11

### Printer

Bambu lab p1s

### How to reproduce

1.-GO to Preview tab after slicing a model
2.-Make sure to have AMS unplugged
3.-Load the external spool filament into the printer.
4.-HIt print and in the filament section you are not allowed to select external spool sources.

[image]

### Actual results

Cannot hit send button on "Send print job" dialog

### Expected results

allow to map into the external spool or ams slots and send to print

### Project file & Debug log uploads

[Speaker_V4_body_131scale.zip](https://github.com/user-attachments/files/19797359/Speaker_V4_body_131scale.zip)

[log.zip](https://github.com/user-attachments/files/19797369/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-04-18:
We changed the mapping rule from V2.0.2 and now you can only map filaments to the correspondin filament type(like what ams did). In your condition, you can set your printer ext to TPU first to make sure the print process.
[image]

XinZhangBBL, 2025-04-24:
@tvixen Can you record a video or GIF that shows the question you are Can you record a video or GIF that shows the question you need? So that we can help you to solve problems

XinZhangBBL, 2025-04-25:
@tvixen 
Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases

[image]

jurgelenas, 2025-04-25:
I think I face similar issue. I am using 2.0.3 and it is not fixed.

[image]

[debug_Fri_Apr_25_09_32_52_71793.log](https://github.com/user-attachments/files/19904035/debug_Fri_Apr_25_09_32_52_71793.log)

[keel clamp.zip](https://github.com/user-attachments/files/19904045/keel.clamp.zip)

```
sudo cat /home/julius/.config/BambuStudio/user/2395974580/filament/Filalab\ PETG.json 
{
    "default_filament_colour": [
        "#000000"
    ],
    "filament_cost": [
        "15"
    ],
    "filament_extruder_variant": [
        "Direct Drive Standard"
    ],
    "filament_settings_id": [
        "Filalab PETG"
    ],
    "from": "User",
    "inherits": "Generic PETG @BBL A1M",
    "name": "Filalab PETG",
    "nozzle_temperature": [
        "250"
    ],
    "nozzle_temperature_initial_layer": [
        "250"
    ],
    "textured_plate_temp": [
        "80"
    ],
    "textured_plate_temp_initial_layer": [
        "80"
    ],
    "version": "1.10.0.27"
}
```

```
sudo cat /home/julius/.config/BambuStudio/user/2395974580/filament/Filalab\ PETG.info 
sync_info = 
user_id = 2395974580
setting_id = PFUS5280d235e3f052
base_id = GFSG99_00
updated_time = 1745588814
```

Even if I create a new filament preset I get same error. New preset json:

```
sudo cat /home/julius/.config/BambuStudio/user/2395974580/filament/Filalab\ PETG\ 2.0.json 
{
    "filament_cost": [
        "15"
    ],
    "filament_extruder_variant": [
        "Direct Drive Standard"
    ],
    "filament_settings_id": [
        "Filalab PETG 2.0"
    ],
    "from": "User",
    "inherits": "Generic PETG @BBL A1M",
    "name": "Filalab PETG 2.0",
    "nozzle_temperature": [
        "250"
    ],
    "nozzle_temperature_initial_layer": [
        "250"
    ],
    "textured_plate_temp": [
        "80"
    ],
    "textured_plate_temp_initial_layer": [
        "80"
    ],
    "version": "2.0.0.87"
}
```

jurgelenas, 2025-04-25:
> [image] Your filament in the spool is PLA and filament type in your slice file is PETG, set the devie's filament type to PETG then you can mapping it right.

This UX is very confusing then! Especially considering the fact that this was not required in the past.

Would it be possible to improve the UX on this one? A lot of users will get stuck on this issue with 2.0 version.

JohnSmithCYM, 2025-04-27:
Thanks for your feedback, and we are considering re-design this UX in the future.

RiverbassX, 2025-04-27:
Consider setting the external spool mapping from the slicer to the device.

Kind of like how the ams works but from the slicer into the external spool

DJF3, 2025-10-28:
Looks like there still are no plans to fix this?

XinZhangBambu, 2025-10-29:
@DJF3 @tvixen Sorry for the confusion.  The reason there is no automatic mapping here is that the material properties in the slicer differ significantly from those in the machine. Initiating a print may cause machine operation issues, poor print quality, etc. So user confirmation is required. Do you have any better suggestions for dealing with this matter?

> [image] Your filament in the spool is PLA and filament type in your slice file is PETG, set the devie's filament type to PETG then you can mapping it right.

XinZhangBambu, 2025-12-30:
@Tamdreah We apologize for any inconvenience caused. Actually, you can set the consumables of the external spool to the actual type, and select the same filament before slicing (the selection can be automatically synced by clicking the consumable drop-down menu).

The word "map" means to map the consumables in the slicer to the actual slots on the machine. Do you have any better word suggestions?

> Hi all, I’m glad I found this thread as I have been driven crazy today with issue. I tried to print with something other than PLA for the first time today (Tpu) and I spent 3 hours search the web and using AI on how to “map” the filament or external spoil. The UI is certainly very confusing when you include the useless message to “map” when I still can’t find a single mention of “map” in any setting menu.
> 
> I finally pressed “ext PLA” box in the “send print job” page and it changed to TPU Ext and I was allowed to print. I still don’t believe I have done this right so if anyone can tell me how to “map” the different filament types to the external spool I would be very greatful