bambulab/BambuStudio #6570: Banbu Studioの要件について

mikio-someya, 2025-04-17:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Printer

Banbu Lab P1S

### How to reproduce

新しいBanbu studio2.0.2の更新のお知らせが来ましたが、ファームウェア要件に
X1シリーズ バージョン01.09.00.00以降。
P1シリーズ バージョン01.08.00.00以降。とあります。

現在P1シリーズの最新バージョンは01.07.00.00ではないでしょうか？

起動するたびにキャンセルしていますので困ります。

### Actual results

更新していませんのでわかりません。

### Expected results

更新しても問題なければ説明文を修正するか、新しいフォームウエアが出るまでお知らせを解除してください。

### Project file & Debug log uploads

[chainbracket v69.zip](https://github.com/user-attachments/files/19797841/chainbracket.v69.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JohnSmithCYM, 2025-04-18:
We are sorry for that and the new firmware will be released later this month. Thanks for your feedback.

BoBoYu-BamBu, 2025-04-28:
@mikio-someya Hello, have your questions been resolved?