bambulab/BambuStudio #6571: Endzeit in BambuStudio fehlt

DessauWetter, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 11

### Printer

BambuLab A1

### How to reproduce

Seit dem Update von BambuStudio sehe ich keine Endzeit des aktuellen Druckauftrag mehr.

### Actual results

[image]

### Expected results

Seit dem Update von BambuStudio sehe ich keine Endzeit des aktuellen Druckauftrag mehr.

### Project file & Debug log uploads

nicht notwendig

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ltownwriter, 2025-04-17:
Das gleiche hier seit dem Update, auch keine Änderung durch die Windows Skalierung 

[image]

lorenz24, 2025-05-25:
Selbes problem bei mir.

[image]

DanBao-Bambu, 2025-11-19:
Hello, please check this issue.

closed 2025-11-19 by DessauWetter