bambulab/BambuStudio #6573: Bambu Studio can see the AMS, but not nothing under Print to turn in on

Coldon, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

WIndows 10

### Printer

X1 Carbon with AMS (original not pro)

### How to reproduce

When setting up a project I can get all the info from the AMS.  But when I go to print, in the SEND PRINT JOB dialog box, there is no option to use the AMS.  So it defaults to the back, printer gives an error because no filament there.  But I can go to Device in Studio and select the color and load it and it works

### Actual results

When setting up a project I can get all the info from the AMS.  But when I go to print, in the SEND PRINT JOB dialog box, there is no option to use the AMS.  So it defaults to the back, printer gives an error because no filament there.  But I can go to Device in Studio and select the color and load it and it works

### Expected results

When setting up a project I can get all the info from the AMS.  But when I go to print, in the SEND PRINT JOB dialog box, there is no option to use the AMS.  So it defaults to the back, printer gives an error because no filament there.  But I can go to Device in Studio and select the color and load it and it works

### Project file & Debug log uploads

When setting up a project I can get all the info from the AMS.  But when I go to print, in the SEND PRINT JOB dialog box, there is no option to use the AMS.  So it defaults to the back, printer gives an error because no filament there.  But I can go to Device in Studio and select the color and load it and it works

### Checklist of files to include

- [ ] Log file
- [ ] Project file

CanoHard, 2025-04-17:
I’m experiencing the same issue. After updating to the latest release, I constantly get errors like "Filament is not loaded" or "The AMS has been disabled for this print..."

Coldon, 2025-04-18:
Exactly. Same thing here. Didn't start until the latest release, though oddly a few days later. I have 2 X1 Carbons and its the same issue on both, so kind of rules out the printer.

caseydawsonjordan, 2025-04-18:
I am having the exact same issue after upgrading firmware to latest

Truth12345, 2025-04-18:
Glad im not alone, same here - I need to preload the filament, otherwise I am getting the "Filament cannot be loaded, check external spool.".

geoffmay, 2025-04-19:
Same here. Please stop beta testing in production.

Torp1lle, 2025-04-19:
Same here. No more changing colors.

XinZhangBBL, 2025-04-21:
@Coldon We sincerely apologize for the inconvenience caused. We are currently resolving the issue and will release a new version as soon as possible.

Coldon, 2025-04-21:
Awesome

pcrui, 2025-04-21:
Same here.I lost several hundred grams of material on two machines.

XinZhangBBL, 2025-04-25:
@Coldon @pcrui @caseydawsonjordan  @geoffmay 
Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases

[image]

geoffmay, 2025-04-25:
Works for me now. Thank you.

Coldon, 2025-04-25:
My AMS is working now.  Still no option to use it or not in the print dialog.  But it always uses it now.  I did switch to LAN mode after the problem started.  I don't know if that makes a difference.

MartinChan3, 2025-04-26:
@Coldon I guess you are trying to do mapping to the specific AMS slot or ext spool. You can give a picture to describe the problem you are encountering.

Coldon, 2025-04-26:
Oh crap, I am good.  Used too before all this happened, you would select to use the AMS or not. Now that the AMS is working, the selection is gone because you moved the ext spool as part of the filament location selection.

I stand corrected, it is indeed fixed.  Thanks and sorry for that.

XinZhangBBL, 2025-04-27:
Thank you.  the issue has been closed since problem solved。

closed 2025-04-27 by XinZhangBBL