bambulab/BambuStudio #6574: setting the cutting plane still not right (follow up #6300)

coat64, 2025-04-17:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

amd ryzen 7 5700
nvidia 4600

### Printer

X1C

### How to reproduce

Although the handling is better and you can set a cutting plane manually without a problem (thanks), the working is still not optimal.

-drop a cube on the plate
-set the angles of all the 3 axis to 30,30,30 (manually or typing in)
-the cutting plane is alright, so no problem yet

### Actual results

But when you rotate the bleu plane (z) the cutting plane is actually not changing but the numbers of all three angles are changing.
Changing the blue plane does not change the cutting plane and is actually useless

Although those numbers may be correct it is not clear for the user. 

In the picture below you can see the cutting plane set by numbers (top)
set manually (middle)
and after changing the blue (Z) angle. changing it change also the other angles. 

[image]

### Expected results

The dial resets itself square to the plate as it does when rotating an object.

[image]

### Project file & Debug log uploads

[cut2.zip](https://github.com/user-attachments/files/19802787/cut2.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

deletedalien, 2025-04-30:
Pretty sure this is linked to this iseue:

Either way neither has been fixed, and its pretty annoying......

https://github.com/bambulab/BambuStudio/issues/6300?fbclid=IwY2xjawJ_WO1leHRuA2FlbQIxMABicmlkETFhR2YyZ3dPdlFQbTFIVlZCAR7mbSzqI6TEV8guoXMmNDhEcRY0j-wWjHsHsk9QeeGh_SMDY1TDaWIriqrzKA_aem_7XRavdaAz7-zAK52lB6aww

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, please update the software version.