bambulab/BambuStudio #6575: Studio Crash

AussieAdelaide, 2025-04-17:
### Bambu Studio Version

2.0.2.5.7

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu H2D

### How to reproduce

Download a file from Makerworld when "slice ok" appears the software crashes.
No problem when importing a STL from Tinkercad the software slices and prints

### Actual results

Software closes before sending the file to the cloud for printing

### Expected results

Should send the file to the printer

### Project file & Debug log uploads

[Pojemnik v2 no logo(5).zip](https://github.com/user-attachments/files/19806001/Pojemnik.v2.no.logo.5.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-18:
We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

remizca, 2025-04-20:
ive been having this issue even on older versions (on A1 Mini printer, had a post here too about it). didnt manage to resolve it. it's annoying as hell and i can't even print properly without the software crashing dozens of times. happens on different files and projects. i even resorted to trying it on vmware but still same problem.

XinZhangBBL, 2025-04-21:
@AussieAdelaide @remizca  We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

remizca, 2025-04-21:
> [@AussieAdelaide](https://github.com/AussieAdelaide) [@remizca](https://github.com/remizca) We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

hello, i actually posted a new thread after replying to this yesterday. i realized i needed to upload the files you guys would probably request

XinZhangBBL, 2025-04-22:
#https://github.com/bambulab/BambuStudio/issues/5900

XinZhangBBL, 2025-04-23:
@AussieAdelaide Halo. We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

XinZhangBBL, 2025-05-19:
Halo, The problem solved or not ? We will close this issue。If not solved, you can reopen the issue

closed 2025-05-19 by XinZhangBBL