bambulab/BambuStudio #6579: custom filament system-level filament preset not available in device tab

boysie123, 2025-04-18:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab p1p

### How to reproduce

Follow method 1 on your wiki page: https://wiki.bambulab.com/en/bambu-studio/create-filament
Confirm the custom filament system-level filament preset not available in device tab

### Actual results

created custom filament system-level filament preset not available in device tab

### Expected results

custom filament system-level filament preset available in device tab

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19809404/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

boysie123, 2025-04-18:
[image]

Missing Elegoo in device tab

[image]

Also they have been through flow dynamics and show there

[image]

weenan, 2025-05-06:
Having the same issue with one of my custom filaments. Printed fine earlier today and now it´s not available on the device tab even though the filament is still available and has the calibration results available too.

I then noticed that it´s missing the preset for the filament but I´m unable to add a preset. (preset was available earlier)

M-A-B007, 2025-05-17:
Facing the same issue repeatedly multiple times. Because I've been experiencing a mysterious loss of custom filaments earlier, I did backups of the filament json files and noticed the following:

1. Yeterday, I've created the custom filament profile. (Also did K-factor calibration.)  
I see two files were created in ...\AppData\Roaming\BAmbuStudio\user\XYXYXYXYXY\filament\base:
  *.info and *.json
The json file contains part:
`"compatible_printers": [ "Bambu Lab X1 Carbon 0.4 nozzle" ]`

2. Today, I've successfully printed the project using the said custom filament. I was able to select it in the AMS slot on the device, and choose the saved K-factor value from yesterday

3. Then I wanted to re-print the same project again with the same custom filament, and noticed that the filament is no longer listed in the AMS slot.5. When I inspected the config files, I found the JSON file had changed:`"compatible_printers": [ ]`

4. See the logs attached. The said filament profile has "Aurapol" in the name.  The dissapearing magic happened somewhen between 2025-05-17 14:00 and 17:00

[log.zip](https://github.com/user-attachments/files/20265405/log.zip)

--------------------------------
Final note: The management of the custom filaments must be improved. It's an imperative. 
I've spent days and days by fixing the missing custom filaments that mysteriously disappeared on their own. 
I've spent days and days by recrerating the tuned custom filaments profiles, after I deleted them by mistake. I was deleting other version of the filament profile and it deleted all the versions including the desired well tuned one.
Really guys, this is a serious trouble and reason to throw the beloved Bambu away and buy the machinery from the competitors again.

M-A-B007, 2025-05-17:
@DanBao-Bambu I can confirm a workaround for the issue. 
After I wrote my previous comment, I added the section `"compatible_printers": [ "Bambu Lab X1 Carbon 0.4 nozzle" ]`
back to the filament profile JSON file. Restarted the printer and restarted the Bambu Studio.
Then I can select the previously lost filament profile in the AMS slot and print with it.