bambulab/BambuStudio #6580: My bambu stdio app doesn't appear on opening

koubidavid, 2025-04-18:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Macbook Pro Sequoia 15.3.2

### Printer

Bambu Lab P1S

### How to reproduce

Click on icon of Bambu studio to open it

### Actual results

modal of loading appears briefly
he menu is available at the top but no window appears anywhere.

### Expected results

Window with bambu studio appears

### Project file & Debug log uploads

[25-04-16 07:10:17.120300][D][2057EC840][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-16 07:10:17.120574][D][2057EC840][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-16 07:10:17.120583][D][2057EC840][IOTC] @ErrCode -12 - Line (13010)
[25-04-16 07:11:01.762438][D][2057EC840][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-16 07:11:01.763038][D][2057EC840][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-16 07:11:01.763061][D][2057EC840][IOTC] @ErrCode -12 - Line (13010)
[25-04-16 07:11:04.242395][D][2057EC840][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-16 07:11:04.242534][D][2057EC840][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-16 07:11:04.242542][D][2057EC840][IOTC] @ErrCode -12 - Line (13010)
[25-04-16 07:15:45.713837][D][2057EC840][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-16 07:15:45.715014][D][2057EC840][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-16 07:15:45.715023][D][2057EC840][IOTC] @ErrCode -12 - Line (13010)
[25-04-17 10:43:49.161808][D][1F1904840][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-17 10:43:49.162260][D][1F1904840][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-17 10:43:49.162267][D][1F1904840][IOTC] @ErrCode -12 - Line (13010)
[25-04-17 11:29:09.567532][D][1F1904840][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-17 11:29:09.568052][D][1F1904840][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-17 11:29:09.568061][D][1F1904840][IOTC] @ErrCode -12 - Line (13010)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-18:
Thank you for your feedback. Could you please provide the screen recording and log files so that we can reproduce and analyze the problem?

koubidavid, 2025-04-18:
I don't have any screens because that is the point. the problem is that
nothing appears. I included all the logs I could gather

bambu123, 2025-04-21:
It should have run to other desktops. We don't know how to reproduce it. The current solution is to delete the configuration file BambuStudio.conf, and if the impact is smaller, only delete the window_mainframe line.
> From: "David ***@***.***>
> Date:  Fri, Apr 18, 2025, 19:07
> Subject:  Re: [bambulab/BambuStudio] My bambu stdio app doesn't appear on opening (Issue #6580)
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> 
  I don&#39;t have any screens because that is the point. the problem is that
<br>
nothing appears. I included all the logs I could gather
<br>

<br>

<br>
<br>

<br>
> Thank you for your feedback. Could you please provide the screen recording
<br>
> and log files so that we can reproduce and analyze the problem?
<br>
<br>
<br>
<br>
> &lt;https://github.com/bambulab/BambuStudio/issues/6580#issuecomment-2815252175&gt;,
<br>
<br>
<br>
<br>
<br>
<br>
<br>
> &lt;https://github.com/bambulab/BambuStudio/issues/6580#issuecomment-2815252175&gt;
<br>
<br>
> Thank you for your feedback. Could you please provide the screen recording
<br>
> and log files so that we can reproduce and analyze the problem?
<br>
<br>
<br>
<br>
> &lt;https://github.com/bambulab/BambuStudio/issues/6580#issuecomment-2815252175&gt;,
<br>
<br>
<br>
<br>
<br>
<br>
<br>

<br>

<br>
--
<br>
—
I don&#39;t have any screens because that is the point. the problem is that
<br>
nothing appears. I included all the logs I could gather
<br>

<br>
<br>
wrote:
<br>

<br>
> Thank you for your feedback. Could you please provide the screen recording
<br>
> and log files so that we can reproduce and analyze the problem?
<br>
<br>
<br>
<br>
> &lt;https://github.com/bambulab/BambuStudio/issues/6580#issuecomment-2815252175&gt;,
<br>
<br>
<br>
<br>
<br>
<br>
<br>
> &lt;https://github.com/bambulab/BambuStudio/issues/6580#issuecomment-2815252175&gt;
<br>
<br>
> Thank you for your feedback. Could you please provide the screen recording
<br>
> and log files so that we can reproduce and analyze the problem?
<br>
<br>
<br>
<br>
> &lt;https://github.com/bambulab/BambuStudio/issues/6580#issuecomment-2815252175&gt;,
<br>
<br>
<br>
<br>
<br>
<br>
<br>

<br>

<br>
--
<br>

VctR477, 2025-11-22:
I have the same problem. 

BambuStudio app doesn't display on the macOS after update to 2.4.0.70

Before this update everything was good.

Apple M2 Pro - macOS Sequoia 15.16.1

I made a video of my screen and upload it to file sharing site.

https://disk.yandex.ru/d/sGNNaIfDg3Rl-g