bambulab/BambuStudio #6582: 识别狭窄内部实心填充 阈值和走线算法发生变化导致 打印件不稳定

shikar, 2025-04-18:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 11.7.10

### Printer

X1C

### How to reproduce

47-49 打印层
[ring.3mf.zip](https://github.com/user-attachments/files/19815030/ring.3mf.zip)

### Actual results

[image]
每个填充只画一个边,而不是完成一个填充,再移动到下一个实心填充,这样会使打印头非常快速的大面积移动,增加了打印面和打印头的摩擦,使打印件被碰的概率增大,所以打印成功率很低, 在 2.0.0.0 之前的版本走线算法就没有问题

### Expected results

应该像2.0.0.0 之前的版本一样完成已螺纹圈,再去打下一个螺纹圈

### Project file & Debug log uploads

[ring.3mf.zip](https://github.com/user-attachments/files/19815073/ring.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

shikar, 2025-04-19:
[image]
V1.10.2.76
版本就是很好

shikar, 2025-04-25:
@XunZhangBambu 怎么样这个问题, 增加了打印时间和提高的失败率

XunZhangBambu, 2025-04-30:
@shikar 在上个版本，我们调整了算法的阈值，导致部分模型效果产生差异，我们会继续优化算法

shikar, 2025-05-04:
@XunZhangBambu 机器固件升级后01.09.00.01, studio 1.10.2.76 不能直接打印了(包括流量测试),表象是不上料

shikar, 2025-06-04:
太棒了 谢谢 @XunZhangBambu