bambulab/BambuStudio #6583: AMS filament sync doesn't update mapping in Support base/interface dropdowns

mshakhmaykin, 2025-04-18:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.3.2

### Printer

X1C

### How to reproduce

1. Change filament in AMS
2. Click "Synchronize filament from AMS"
3. See that types and colors are updated in "Project filaments" list.
4. Go to Global -> Support and click one of the two "Filament for Supports" dropdowns

### Actual results

See that the types and colors in "Filament for Supports" are not updated

### Expected results

See that the types and colors in "Filament for Supports" are updated and match "Project filaments" list.

### Project file & Debug log uploads

I don't think it's relevant to a project.

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-21:
Hello, I did not reproduce your issue~
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-04-21:
Hello, we have identified the issue. Thank you for your feedback

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

closed 2025-04-25 by Haidiye00