bambulab/BambuStudio #6587: Cant send prints on 2.0.2 because of non-existant AMS?

ThreeHats, 2025-04-19:
### Bambu Studio Version

2.0.2

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

[image]

With the new update, I cannot send prints. It seems like it want me to select a filament with the AMS, but I have never had one.

### Printer

Bambu P1P

### How to reproduce

Try to print anything without an AMS

### Actual results

[image]

### Expected results

To be able to print

### Project file & Debug log uploads

I did not get the logs before downgrading to 2.0.0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ZenBudd, 2025-04-19:
I have same problem. Older project from Bambu Studio ver. 1 x x  saved at SD card cant print by Bambu Studio ver. 2 x x.
Manually chose AMS filament possition cant help. Printer P1S only prepare temperatures and dont trying to load filament from AMS.
I have many projects at SD card and i dont have a time to making all by Bambu Studio ver 2 x x. I hope, they repair it. Just i go back to ver 1 x x.

JohnSmithCYM, 2025-04-20:
(As #6540 ) We are sorry we didn't make it clear on release note that we cancel "Use AMS" on X/P/A(like H2D did) and you can select ext spool just click the mapping window to map to ext manually.

[image]

ThreeHats, 2025-04-20:
[image]

That is not an option. There is no way to select *any* filament

ThreeHats, 2025-04-20:
[debug_network_Sun_Apr_20_14_46_58.log](https://github.com/user-attachments/files/19826901/debug_network_Sun_Apr_20_14_46_58.log)
[debug_Sun_Apr_20_14_46_58_3964.log](https://github.com/user-attachments/files/19826900/debug_Sun_Apr_20_14_46_58_3964.log)
[debug_Sun_Apr_20_14_46_30_24416.log](https://github.com/user-attachments/files/19826898/debug_Sun_Apr_20_14_46_30_24416.log)
[debug_network_Sun_Apr_20_14_48_50.log](https://github.com/user-attachments/files/19826897/debug_network_Sun_Apr_20_14_48_50.log)
[debug_Sun_Apr_20_14_48_33_26280.log](https://github.com/user-attachments/files/19826899/debug_Sun_Apr_20_14_48_33_26280.log)

Log files (without the .0 file extension? Why are all of my logs "filename.log.0"?)

ThreeHats, 2025-04-20:
[image]

[image]

Ok, I figured it out. The filament selected on the device screen need to be changed to match the expected type. This should be made clear in the UI, and maybe an option to change it when sending the print job

XinZhangBBL, 2025-04-21:
@ThreeHats Sorry for the problem. we will make it clear in the UI and release the version soon

ZenBudd, 2025-04-21:
Select external spool is not a solution, if i have multi color project. I set manualy correct filament possitions in AMS and printer is only heating hotend and hotbed. AMS is stopped and dont work.

XinZhangBBL, 2025-04-25:
@ZenBudd Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases
[image]

ZenBudd, 2025-04-25:
> [@ZenBudd](https://github.com/ZenBudd) Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases [image]

Hello, this ios not solution for my issue.
All my project at  SD card is not statrt printing with Bambu Studio ver. 2. 
If i chose Device - Storage - Model and chose project created in Bambu Studio ver 1.x.x. i see this picture:

[image]

Next i chose propper filament from AMS:

[image]

And click on send.
Now printer P1S is heating hot end and hot bed, but AMS id not work anything....Bambu Studio say filament loading, AMS do nothing...

If i downgrade to Bambu Studio ver 1.x.x. all works good.

[image]

JohnSmithCYM, 2025-04-27:
@ZenBudd Thanks for your feedback, can you offer two things for us:
- This gcode file example.
- (Reproduce this problem again) and send your log file for us to check.