bambulab/BambuStudio #6588: Timelapse Grayed Out

markeverettbeck, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu p1s

### How to reproduce

1. Select Slice Plate
2. Select Print Plate 

### Actual results

Grayed Out Timelapse defaulting to "Off"

[image]

### Expected results

Timelapse to be set to on, or at least able to turn it on.

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/user-attachments/files/19819626/New.Compressed.zipped.Folder.zip)

This happens with every project file now, and the option enables if I turn the p1s off and on again, but shortly after slicing the plate again it grays out again

### Checklist of files to include

- [x] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-20:
Can you upload your studio log for us to find out the real question? You can follow https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

XinZhangBBL, 2025-04-21:
Could you please move the cursor above the text “Timelapse”。 There will be toolip to inform why the timeplase option is disabled. @markeverettbeck

closed 2025-05-19 by XinZhangBBL