bambulab/BambuStudio #6589: 2.02.57 AMS lite problem with bambulab A1

yoozoon, 2025-04-19:
### Bambu Studio Version

2.02.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1

### Printer

bambulab A1 with AMS lite

### How to reproduce

Since 2.02.57 Bambu Studio was install, I have not been able to communicate correctly with my A1’s AMS lite with any new slices. It seems to always be looking for an external spool or uses the wrong spool. Only fix has been to roll back to 2.01.50. Now I can slice models and the AMS Lite works correctly.

### Actual results

while starting a multicolor print the AMS lite is not working it does not feed the filament but is expecting an external roll

### Expected results

the AMS lite should feed the filament to the extruder

### Project file & Debug log uploads

[debug_Fri_Apr_18_16_24_27_12016.log.0.zip](https://github.com/user-attachments/files/19820066/debug_Fri_Apr_18_16_24_27_12016.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Truth12345, 2025-04-19:
This has already been reported here:
https://github.com/bambulab/BambuStudio/issues/6573

Its the same issue, not only with ams lite but generally on 2.02.57

XinZhangBBL, 2025-04-21:
@Coldon We sincerely apologize for the inconvenience caused. We are currently resolving the issue and will release a new version as soon as possible.

XinZhangBBL, 2025-04-25:
@yoozoon Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases

[image]

yoozoon, 2025-04-25:
> [@yoozoon](https://github.com/yoozoon) Halo. The new version has been released, please check if your problem has been solved. https://github.com/bambulab/BambuStudio/releases
> 
> [image]

Thank you very much, first tests look good so far.

XinZhangBBL, 2025-04-25:
@yoozoon Thank you for your feedback. Issue closed.

closed 2025-04-25 by XinZhangBBL