bambulab/BambuStudio #6590: Unexpected slicing results with old versions, with .3mf files created in 2.0.2.

jcollasius, 2025-04-19:
### Bambu Studio Version

2.0.2

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 [Version 10.0.26100.3775]

### Printer

Bambu Lab P1P

### How to reproduce

I had a user using Bambu Studio 1.9. He used my print profiles, created with Bambu Studio 2.0.2 and hat massiv slicing problems. I use a modifier to set top and bottom layers to 0, wall thickness to 0 and change the infill type and density.
Apparently Bambu Studio 1.9 ignores any settings made to object modifiers.

Go to https://makerworld.com/en/models/1318907-no-spill-ams-desiccant-boxes-round-hygro#profileId-1354659
Download print profile
Open profile in Bambu Studio 1.9

### Actual results

Have solid bottom.

[image]

### Expected results

[image]

### Project file & Debug log uploads

[B+-+Top+Aligned+-+No+spill+AMS+silica+gel+desiccant+boxes.zip](https://github.com/user-attachments/files/19820308/B%2B-%2BTop%2BAligned%2B-%2BNo%2Bspill%2BAMS%2Bsilica%2Bgel%2Bdesiccant%2Bboxes.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-04-30:
Old versions of Studio may not be compatible with files generated by new Studio. Please update to the new version.

jcollasius, 2025-04-30:
Is the newest Studio version compatible with .3mf files generated in old Studio versions?
I.e. I would use Bambu Studio 1.9 to create my .3mf files for MakerWorld and anyone could use them without problems?