bambulab/BambuStudio #6591: Fuzzy skin also on inside when selected “Contour”

eriklins, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.3.2

### Printer

Bambu Lap P1S (likely any other as well)

### How to reproduce

1) Open fuzzy_skin_contour_issue.3mf with Bambu Studio.
2) Check under "Other" that Fuzzy Skin is enabled and set to "Contour".
3) Slice it.
4) See the Fuzzy Skin flowing through a hole in the outer wall to the inside of the model.

### Actual results

When enabling Fuzzy Skin "Contour" and there is a hole in the outer wall, the Fuzzy Skin sort of flows through the hole to the inside of a model.

### Expected results

When enabling Fuzzy Skin "Contour" it should only add Fuzzy Skin to the outside of a model independent of any holes in outer walls.

### Project file & Debug log uploads

[fuzzy_skin_contour_issue.3mf.zip](https://github.com/user-attachments/files/19820658/fuzzy_skin_contour_issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-05-23:
Thanks for the report. We will add some tools to resolve this issue and release them by the end of June.

QingZhangBambu, 2025-07-21:
for now, you can add modifier to block it.