bambulab/BambuStudio #6593: bambu lab studio crashes when you click "preview"

zrex-3mf, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 version 22h2 (OS Build 19045.5737)

### Additional system information

CPU: Intel(R) Core(TM) i7-6800K CPU @ 3.40GHz
Memory: 32GB
Display Card: NVIDIA GeForce GTX 1080 Ti

[image]
[image]
[image]
[image]

### Printer

Bambu Lab X1C

### How to reproduce

1. click bambu studio
2. click on model to see/prepare the model in the "prepare" tab on studio
3. click "preview" ....CRASH... sometimes with pop ups sometimes without

[image]
[image]
[image]
[image]

### Actual results

Studio keeps crashing when i click "preview", to slice any model so i can print it. the the studio just vanishes when i click "preview" or sometimes it also pops up these messages and then vanishes on it own as well. not allowing me to "preview", which then doesn't allow studio to slice so then i cannot print because the studio doesn't slice at all, the studio just vanishes when i click "preview".
"A fatal error occurred: "Access violation"  Please save project and restart the program."  
"Invalid spacing supplied to Flow+with spacing0"

[image]

[image]

[image]

[image]

### Expected results

being able to slice any model and print any model.

### Project file & Debug log uploads

[wall plug keeper .370 cable dia zm CRASH.zip](https://github.com/user-attachments/files/19821713/wall.plug.keeper.370.cable.dia.zm.CRASH.zip)

[glock pen CRASH.zip](https://github.com/user-attachments/files/19821714/glock.pen.CRASH.zip)

[Flying Sea Turtle CRASH.zip](https://github.com/user-attachments/files/19821715/Flying.Sea.Turtle.CRASH.zip)

[disk thrower toy CRASH.zip](https://github.com/user-attachments/files/19821716/disk.thrower.toy.CRASH.zip)

[bambu-studio.exe.8816.zip](https://github.com/user-attachments/files/19821730/bambu-studio.exe.8816.zip)

[bambu-studio.exe.2268.zip](https://github.com/user-attachments/files/19821773/bambu-studio.exe.2268.zip)

[bambu-studio.exe.12856.zip](https://github.com/user-attachments/files/19821775/bambu-studio.exe.12856.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

zrex-3mf, 2025-04-21:
anyone there? @XunZhangBambu  @JohnSmithCYM  
it really sucks not being able to print what i want to print and need to print. any answers or fixes? it may be getting worse, studio wont slice a benchy now either..

zrex-3mf, 2025-04-21:
@QingZhangBambu @XunZhangBambu @johnsmithCYM 

any progress?

zrex-3mf, 2025-04-23:
I am still getting the error  “A fatal error occurred: "Access violation"  when I choose PLA to be the "Support/raft interface" under "Filament for Supports", which is what I have always done. 
Why won't the Bambu studio slicer allow me to do that now? its stopping me from doing what I want with the materials I want. Why is this happening all of a sudden after the most recent update? I need to be able to print PLA with PETG and visa versa for dissimilar interfacing supports.

zrex-3mf, 2025-04-23:
@XunZhangBambu 
am I missing something, do i have a wrong preset, or did the most recent update last week change something without notification? that's disallowing me to slice a model with dissimilar material for support interface?

XunZhangBambu, 2025-05-15:
@zrex-3mf  Sorry for the delay response. Have you tried with our latest version ?  I didn't  encounter crash when setting support interface filament.  This may be because your version of the studio is already in an abnormal state after an error occurred.