bambulab/BambuStudio #6594: H2D Process Profiles Appear to Be Incorrect

hifihedgehog, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

This is less an issue with Bambu Studio and more to do with the built-in process profiles. The H2D process profiles have serious issues and are not production ready. First, the normal printing acceleration is 8000 mm/s² for the normal printing profiles whereas that same setting is 4000 mm/s² for other profiles such as balanced strength. On your other printers, the normal printing acceleration is consistent from process profile to process profile.  The other issue is the process profiles do not match their desired application. For example, the balanced strength profile, a profile a user would expect to offer highly enhanced strength like the strength profile on the X1, P1 and A1 series, has just 2 wall loops and 15% infill. On previous generations, the strength profiles had 6 wall loops which aligns with guidance you give users while the new strength profile on H2D runs contrary to this (see [here](https://www.youtube.com/watch?v=FRLympzHsWY) and [here](https://x.com/BambulabGlobal/status/1636433235201720322)) and established norms of 3D printing design. This needs to be fixed!

### Printer

Bambu Lab H2D

### How to reproduce

See above.

### Actual results

See above.

### Expected results

See above.

### Project file & Debug log uploads

See above.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hifihedgehog, 2025-04-19:
# Consistency of normal printing acceleration of process profile on other Bambu Lab printers versus H2D

## A1 mini

[image]

[image]

## X1 Carbon

[image]

[image]

## H2D

[image]

[image]

hifihedgehog, 2025-04-19:
# Strength process profile regression in H2D

[image]

[image]

Following the process profiles of your prior printers, your own recommendations, and established standards for print strength in additive manufacturing, this is not a strength profile.

hifihedgehog, 2025-04-19:
Users purchasing a flagship, especially the H2D which was delayed to ensure everything was ready for launch and which carries a significantly high premium price tag, should not have to deal with out-of-the-box process profile shortfalls like this. Minimally, we would expect the profiles from your previous printers to have been ported over with the necessary changes to accommodate the larger design. Instead, the process profiles, outside of the normal printing profile, by every indication at closer examination, appear to be rushed, untested, and unrevised. I share this as a proud Bambu Lab customer and would hope will see these shortfalls as a learning opportunity to improve and live up to your brand's reputation.

hifihedgehog, 2025-04-19:
Users purchasing a flagship, especially the H2D which was delayed to ensure everything was ready for launch and which carries a very high premium price tag, should not have to deal with shortfalls in the out-of-the-box process profile like this. They should generally be mostly tuned and not be incorrect and inconsistent like this. Minimally, customers would expect the profiles from your previous printers to have been ported over with the necessary changes to accommodate the larger design, but we see no effort here to provide purchasers this experience. Instead, the process profiles, outside of the normal printing profile, by every indication at closer examination, appear to be rushed, untested, and unrevised. I share this constructive criticism as a proud Bambu Lab customer because I would hope you would see these shortfalls as a learning opportunity to improve and strive to live up to your brand's reputation.

hifihedgehog, 2025-04-19:
Customers purchasing a flagship product, particularly the H2D, which was delayed to ensure readiness for launch and carries a significant premium price tag, should not encounter deficiencies in the out-of-the-box process profiles as outlined above. The process profiles should generally be well-optimized and not exhibit errors or inconsistencies of this nature. At a minimum, customers would expect the profiles from previous printers to have been adapted with the necessary modifications to accommodate the larger design, yet there appears to be no effort to deliver this experience. Instead, the process profiles, apart from the standard printing profile, upon closer examination, seem rushed, untested, and unrefined. This constructive criticism is offered as a loyal Bambu Lab customer in the hope that these shortcomings will be viewed as an opportunity for growth and improvement, aligning with the high standards of the brand's reputation.

xsimtomx, 2025-04-21:
While minor, but kinda related: the descriptions for each process profile are exactly the same for all (unlike on other printers).

[image]

hifihedgehog, 2025-04-21:
> While minor, but kinda related: the descriptions for each process profile are exactly the same for all (unlike on other printers).
> 
> [image]

This is another point and another issue I found in digging though the .jsons that I neglected to mention. Even the "balanced strength" profile carries that same description (i.e. "It has a normal layer height (...) suitable for most printing cases") the balanced profile even though they should be descriptive. This is a major sticking point since this is not a tinkerer printer but a printer that a user should be able to use to just "print and forget." I am surprised how none of the reviews really commented on these serious profile shortcomings.

Another issue I will be eventually reporting is how the print time ends up being sometimes slower with high flow nozzles than the standard flow nozzles. I have yet to figure out what is causing this behavior as it only happens with certain models. I will be providing a model (or hopefully models) when I post the issue so they can hopefully figure out what is wrong with their slicing algorithm that would cause it to have a high flow nozzle operate more slowly than the standard flow nozzle.

XunZhangBambu, 2025-04-30:
@hifihedgehog  For the H2D , the extra fine process can be considered as a high-quality configuration of the old printer, aimed at achieving the best printing results. Therefore, the speed differs compared to the standard configuration.

hifihedgehog, 2025-05-03:
@XunZhangBambu, I appreciate the response. However, I mentioned _**nothing**_ about extra fine in my comments or report. If necessary, please push this up the chain so this can be properly vetted. What you are conveying to users makes no sense and is totally inconsistent with previous configurations. The strength profile, for example, is totally wrong and is the example I actually gave above.

That being said, I will address your direct remark. As for extra fine which you are the only one who mentioned it, there are some interesting choices on the speeds compared to X1 Carbon and even A1. The speed and acceleration there are significantly slower. If it is meant to be a more quality version of extra fine, then name it as such with a label like "Extra Fine Quality." Silently reducing speed without letting your customers know is totally unacceptable for a flagship product. Poll owners and they will tell you this.

Kemmey, 2025-05-03:
As completely new to the bambulab ecosystem, with both an a1 mini and h2d - and access to several P1s, this totally confuses my out too - and I fully ecco everything @hifihedgehog is saying. There needs to be a full set of profiles developed for the h2d, mirroring every other bambulab printer. This seems an impossible oversight - like - did you not internally fully test the 3D printing aspects of this printer?!?

hifihedgehog, 2025-05-03:
> As completely new to the bambulab ecosystem, with both an a1 mini and h2d - and access to several P1s, this totally confuses my out too - and I fully ecco everything [...]

Design- and hardware-wise, I believe the H2D was thoroughly vetted. Their engineers are ex-DJI employees and mechanical kinematic design is their typical simple elegance that makes overbuilt attempts like Creality's K2 series look like amateur hour. The issue this round that I am seeing and needs major thought so this is not repeated is the H2D's profiles are clearly rushed. I have been poking around and the issue is they are not using the base dual-head profile correctly. In previous printers, there is a base profile that forms the speeds and accelerations of every other profile. For H2D, they created a new dual-head base profile that serves as the foundation for the other default profiles. However, for some unexplainable reason, that profile isn't either fully working or the underlying code isn't ready. Really sad.