bambulab/BambuStudio #6595: Tree Supprts dont generate support interface and makes the support fuse to the piece

deilvad0311, 2025-04-19:
### Bambu Studio Version

2.2.0.57

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

A1 mini

### How to reproduce

Activate Tree supports, and slice

### Actual results

the tree supports dont generate any support interface and makes it fuse to the piece

### Expected results

generate support interace to be able to remove supports

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-21:
Thank you for your feedback. Please provide the 3mf file and screenshots to describe your issue so that we can reproduce and analyze it.

NothFie, 2025-05-06:
@deilvad0311 Thanks for your feedback. Due to the lack of files and images, I can only guess the issue you are facing is that there is no interface for SharTail areas. It's a bug and it's fixed in the latest version.