bambulab/BambuStudio #6596: Acceleration values ​​have no effect on the H2D printer

Empire4191, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

In Bambu Studio, the acceleration values ​​have no effect on the H2D printer.
This means that the print time is exactly the same, regardless of whether I set it to 5000 mm/s² or 5 mm/s².
This is likely a bug in the software/firmware. Is this already known?

### Printer

H2D

### How to reproduce

.

### Actual results

.

### Expected results

.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-21:
@Empire4191 OK. Have your problem been resolved?

Empire4191, 2025-04-21:
Yes, thank you