bambulab/BambuStudio #6597: Avoid crossing walls gives up when using 0.6 nozzle but not on 0.4

boysie123, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

Bambu Lab P1S

### How to reproduce

using the provided project slice using 0.6
check the top section for travel path
Using the same project slice with 0.4
check the top section for travel path

### Actual results

when using 0.4 nozzle the path is correct, avoiding crossing the outer wall when doing infill to infill
when using 0.6 nozzle the path is incorrect, crossing over the outer wall causing a failed print and artifacts to appear.

### Expected results

the tool path correctly avoids crossing the outer wall like it does when using 0.4 nozzle and like it does for the rest of the model lower down when using 0.6, it seems to give up trying to avoid the outer wall for some reason like a limit has been reached and it cant be bothered any more :)
It is a complex model but still I wasn't expecting this. 

### Project file & Debug log uploads

[Japandi Vase 0.24.zip](https://github.com/user-attachments/files/19822532/Japandi.Vase.0.24.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

boysie123, 2025-04-19:
[image]

boysie123, 2025-04-19:
as per: https://github.com/SoftFever/OrcaSlicer/issues/5217#issuecomment-2808307090

it looks like you can work around it by forcing a retraction which should be unnecessary

XunZhangBambu, 2025-04-30:
@boysie123  Our developers have started analyzing this issue. Thank you for the feedback.