bambulab/BambuStudio #6598: bambu studio

Duudeeee, 2025-04-19:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11

### Printer

X1C 

### How to reproduce

Im printing PLA and try to change speed fans and dont let me acess that area

[image]

### Actual results

dont work!

### Expected results

have to work 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Duudeeee, 2025-04-19:
the handystudio dont let me change the FAN SPEED to, its show allways the bad info , shows 30% fan part,  and dont let me change the others fans to the % i want

Duudeeee, 2025-04-19:
i forget to say that im on BETA firmware.

BoBoYu-BamBu, 2025-04-21:
@Duudeeee Sorry, may I ask what version of the printer firmware you have installed?

Duudeeee, 2025-04-21:
> [@Duudeeee](https://github.com/Duudeeee) Sorry, may I ask what version of the printer firmware you have installed?

Beta firmware

shuizhiguo, 2025-04-22:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

guywholikestobreakthings, 2025-04-25:
I have this same issue!!!!

OzButcher78, 2025-05-11:
I have same issue.... can't adjust fans using bambu studio on the X1C

i3dm, 2025-05-11:
same issue here too