bambulab/BambuStudio #6600: Print progress current layer stuck at zero

Scanjo, 2025-04-20:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

Bambu Lab P1S

### How to reproduce

[image]

### Actual results

Doesn't show current layer. Always shows "0".

### Expected results

It should show the current layer being printed.

### Project file & Debug log uploads

[LadyBug.zip](https://github.com/user-attachments/files/19823388/LadyBug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Scanjo, 2025-04-20:
It wouldn't let me type in the field about how to reproduce. Only send a file so I sent a pic showing the behavior. It happens on all my prints. Just send a job to the printer and it always shows layer zero of total layers. 

P1S Firmware 1.6.1.2

JohnSmithCYM, 2025-04-25:
Can you offer your printer's gcode file for us to check if the specific gcode was wrong? And can you reproduce it this days?

Scanjo, 2025-04-27:
Hello JohnSmithCYM,

I created a new job to be sure it's doing it all the time. Attached is another screen shot and the exported Gcode for this print.

Thursday, April 24, 2025, 10:39:21 PM, you wrote:

> JohnSmithCYM left a comment (bambulab/BambuStudio#6600)

> Can you offer your printer's gcode file for us to check if the specific gcode was wrong? And can you reproduce it this days?

-- 
Best regards,
Scott Johnson

Scanjo, 2025-04-27:
Hello JohnSmithCYM,

The finish time is in the wrong place too, but since it seems to reflect the time to finish accurately I just consider this cosmetic. The layer bug is showing bad info so it should really be fixed.

Thursday, April 24, 2025, 10:39:21 PM, you wrote:

> JohnSmithCYM left a comment (bambulab/BambuStudio#6600)

> Can you offer your printer's gcode file for us to check if the specific gcode was wrong? And can you reproduce it this days?

-- 
Best regards,
Scott Johnson

JohnSmithCYM, 2025-04-27:
Thanks for your feedback, we have received your GCode and our RD is analyzing it.

gaolegao-lx, 2025-04-28:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Scanjo, 2025-04-30:
Hello gaolegao-lx,

I can't send a log yet. If I understand correctly, to send a log I must start a ticket and then proceed with the print job. Unfortunately I'm in the middle of a big print and I'll have to wait till it's done. Projected 2 days provided I don't have issues with the AMS (tangles and not wanting to load lately). I will upload a log for you as soon as this is done.

Sunday, April 27, 2025, 8:14:49 PM, you wrote:

> gaolegao-lx left a comment (bambulab/BambuStudio#6600)

> **Please upload log through Handy APP, and give us your ticket number**
> ***
> ***Upload log files through Bambu Handy***
>      Log files from the Bambu printer can be uploaded via Bambu Handy over the [...]

-- 
Best regards,
Scott Johnson

Scanjo, 2025-05-03:
Hello gaolegao-lx,

Ticket #US250503849001

Sunday, April 27, 2025, 8:14:49 PM, you wrote:

> gaolegao-lx left a comment (bambulab/BambuStudio#6600)

> **Please upload log through Handy APP, and give us your ticket number**
> ***
> ***Upload log files through Bambu Handy***
>      Log files from the Bambu printer can be uploaded via Bambu Handy over the [...]

-- 
Best regards,
Scott Johnson

Scanjo, 2025-05-04:
I figured out the issue. You can stopp looking. It is NOT a BBS problem. I have custom gcode installed from the SteinWipe project and THEY had a bug.

This can be closed. Sorry I inconvenienced you.

closed 2025-05-04 by Scanjo