bambulab/BambuStudio #6603: Makerworld files do not load

Kebratos, 2025-04-20:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1

### How to reproduce

1. Go to makerworld
2. Choose a file
3. Click on open in bambu studio
4. see the error, it will never load de file

### Actual results

[image]

### Expected results

It should open the file

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-21:
Thank you for your feedback. Could you please provide the details of the model connection that you are unable to access?

Haidiye00, 2025-04-21:
Hello, can you download the 3mf version of Makeworld and then open it with the software? If you still can't open it, can you share this 3mf? Thank you

Haidiye00, 2025-06-03:
Hello, please use the latest version of the software. Thank you~
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59

[image]
If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software