bambulab/BambuStudio #6605: bambu studio password reset is BROKEN

sparoad, 2025-04-20:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.4.1 (24E263)

### Additional system information

HOW do I reset my Password in Bambu Studio, when it takes more time than they allow for their reset pasword email to arrive here in New Zealand from wherever they send it from. Even when trying to reset it via GitHub, that has the same SLOW “feature”!

### Printer

bambu a1

### How to reproduce

first try
1. click on login/register
2. it won't acknowledge my email address
3. GIVE UP

second try
1. click on login/register
2. click on create your account
3. click on Send verification code
4. AFTER WAITING FOR SEVERAL HOURS give up
5. use with logging in seems to work OK

### Actual results

nothing

### Expected results

being able to create a new account
or
being able to login

### Project file & Debug log uploads

nothing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-29:
@sparoad Hello, have your questions been resolved?

sparoad, 2025-05-07:
no. it is still a problem