bambulab/BambuStudio #6609: Top Surface Pattern Issues - v2.0.2.57

crockerg, 2025-04-20:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Mac 15.3.2 (24D81)

### Printer

X1C

### How to reproduce

Attached project shows example of the Hilbert Curve not be applied to the top of the Test Text lettering. This was supported in previous versions of BambuSlicer

### Actual results

[image]

### Expected results

I would expect the Hilbert Curve to be applied to the entire top surface of the letters. This is how it worked in previous versions.

### Project file & Debug log uploads

[Hilbert Test.3mf.zip](https://github.com/user-attachments/files/19825612/Hilbert.Test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-04-24:
which version could slice it successfully? i will check and compare

crockerg, 2025-04-24:
V1.10.2.76 for sure works.

But this is recent and I think earlier versions of v2 worked fine

QingZhangBambu, 2025-05-23:
sorry for reply it later.  i've checked v1.10.2.76. you text is stll to small to generate Hilbert curve.

[image]