bambulab/BambuStudio #6610: Problems sending the file trough the cloud to the printer.

3dPrintUrmind, 2025-04-20:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OSX ventura 13.7.5

### Printer

AI with AMS lite

### How to reproduce

just sending a file to the printer
same network
the error that pop 
 0500-400B 312000

### Actual results

problems with network connections 

### Expected results

been able to send the file to print

### Project file & Debug log uploads

don't know how

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-21:
Sorry, it might be that the network was not stable during that period. Could you try again now? If it still doesn't work, please provide screenshots and logs.

3dPrintUrmind, 2025-04-21:
Hi,fixed, i tried with a hot spot and worked, has to be the T mobile
router that has the port closed or its has been blacklisted by it. I will
contact t mobile, thanks, can you tell me the ports that the cloud service
use for the printer? so i can fix this with tmobile. thanks
Ariel M.Pagan Mejias
AMSA 161 G 81st RSC
C & E Electronics Technician(RET)
Ft.Buchanan PR.
"When the pin is pulled,Mr.Grenade is not our friend anymore" (US Special
Forces.)

Confidentiality Note:
This message is intended for use
only by the individual(S) or entity to which it is addressed
and may contain information that is privileged,
confidential, and exempt from disclosure under applicable
law.  If the reader of this message is not the intended
recipient or the employee or agent responsible for
delivering the message to the intended recipient, you are
hereby notified that any dissemination, distribution or
copying of this communication is strictly prohibited.  If
you have received this communication in error,  please
contact the sender immediately and destroy the material in
its entirety, whether electronic or hard copy.  Thank you.

BoBoYu-BamBu, 2025-04-21:
@3dPrintUrmind OK. So, can we close this order now?

3dPrintUrmind, 2025-04-21:
Yes please
 and thanks for your fast reply.
Ariel M.Pagan Mejias
AMSA 161 G 81st RSC
C & E Electronics Technician(RET)
Ft.Buchanan PR.
"When the pin is pulled,Mr.Grenade is not our friend anymore" (US Special
Forces.)

Confidentiality Note:
This message is intended for use
only by the individual(S) or entity to which it is addressed
and may contain information that is privileged,
confidential, and exempt from disclosure under applicable
law.  If the reader of this message is not the intended
recipient or the employee or agent responsible for
delivering the message to the intended recipient, you are
hereby notified that any dissemination, distribution or
copying of this communication is strictly prohibited.  If
you have received this communication in error,  please
contact the sender immediately and destroy the material in
its entirety, whether electronic or hard copy.  Thank you.

closed 2025-04-21 by BoBoYu-BamBu