bambulab/BambuStudio #6613: Timelapse

TabouretLudique, 2025-04-20:
### Bambu Studio Version

2.0.2°57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

Bambu Lab A1

### How to reproduce

When I change the timelapse mode from smooth mode to traditional mode, there is no change in the slicing result.

### Actual results

No change.

### Expected results

The slicing should be different depending on the selected timelapse mode.

### Project file & Debug log uploads

[image]
[image]

[1x-engineblockv6.zip](https://github.com/user-attachments/files/19827017/1x-engineblockv6.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-04-21:
@TabouretLudique Because in this case, the gcode paths generated by the traditional mode and the smooth mode are very similar.  So the time is the same, which is within the expected range.

TabouretLudique, 2025-04-21:
Thank you for your reply. In a traditional timelapse mode, the nozzle is not supposed to hide between photos. Do you know if it's possible to disable the timelapse feature or to modify the timelapse code?