bambulab/BambuStudio #6615: Wipe tower moves in the way when replacing an object on the plate.

coat64, 2025-04-20:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

Ryzen 5700g
nvidia 4600

### Printer

H2d

### How to reproduce

See project
Move object in Prepare-tab and prime tower jumps to another place and in the default square look. also tower is limited in placing.
Slice and it gives a warning for conflicting G-code paths
Return to prepare and now the prime tower has its real shape with shapes and can be placed whwre you want and out of the way.
Now you have to slice it again.

[image]

[image]

[image]

[image]

### Actual results

As described above, prepare - cube in the way, slice, prepare and move tower > slice > print 

### Expected results

The wipe tower is free to move the first time.

### Project file & Debug log uploads

[onderbouw_achterkant1.zip](https://github.com/user-attachments/files/19827600/onderbouw_achterkant1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-04-30:
@coat64  Thank you for reporting this issue, we will try to improve it.

zhimin-zeng-bambulab, 2025-06-05:
@coat64  The new version ([v02.01.00.59](https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59)) has been optimized for this problem, you can have a try.

closed 2025-06-26 by zhimin-zeng-bambulab