bambulab/BambuStudio #6616: Bambu Studio crashed with PETG support interface layer

brianharwell, 2025-04-21:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1C and H2D

### How to reproduce

[Bridge Test.zip](https://github.com/user-attachments/files/19827637/Bridge.Test.zip)

Slice the attached file.

Manual Steps to Reproduce:

1. Enable supports
2. Type: Normal (Auto)
3. Support Interface: Select a PETG filament
4. Top Interface Layers: 1

### Actual results

Application crashed with an Access Violation

[image]

### Expected results

Model should be successfully sliced.

Model can be successfully sliced if `Support Type` is changed to `Tree (Auto)` or `Top Interface Layers` is set to `1`

### Project file & Debug log uploads

Log upload failed, I will try to upload it as a comment

### Checklist of files to include

- [x] Log file
- [x] Project file

brianharwell, 2025-04-21:
I can't upload the memory dump. It errors

```
<!-- Failed to upload "debug_network_Sun_Apr_20_20_13_33.log.zip" -->
```

NothFie, 2025-04-22:
@brianharwell Thanks for your feedback. It's a BUG that studio crashes when Top_Interface_Layers is set to 1 and it will be fixed in the next version soon. I deeply apologize for any inconvenience caused to you.

NothFie, 2025-04-30:
The issue is fixed

closed 2025-04-30 by NothFie