bambulab/BambuStudio #6618: 更新后，CPU利用率随时都保持较高

snxiaomoon, 2025-04-21:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Printer

A1MINI

### How to reproduce

直接打开就重现了

### Actual results

[image]

### Expected results

之前不会这样的

### Project file & Debug log uploads

没有

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-21:
Hello, we have fixed this issue and the next version will include it~

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

closed 2025-04-25 by Haidiye00