bambulab/BambuStudio #6621: External spool not seen

Westoz123, 2025-04-21:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS sequoia 15.3.2

### Printer

X1C

### How to reproduce

I have been happily printing to my AMS and to the external spool when needed, but now it doesn't see the external spool even when it is loadfed on the printer screen.
I have rebooted the Mac. rebooted the printer, disconnected the AMS and rebooted the printer. Reconnected the AMS, rebooted the printer, reloaded the external spool filament. Synchronised the filaments to the Mac. But the filaments in the AMS are different from those shown on the mac and when I go to print the screen does not show the  turn off AMS dialogue box like it did yesterday.
Very frustrating. what can I do ?

### Actual results

Nothing changes

### Expected results

It should work like it has for ages

### Project file & Debug log uploads

Help

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-21:
Thank you for your feedback. Could you please provide a screenshot and a video recording to describe your problem?

Westoz123, 2025-04-22:
I just made a short video.
<https://youtu.be/r2d71dEQYNo>
22 April 2025<https://youtu.be/r2d71dEQYNo>
youtu.be<https://youtu.be/r2d71dEQYNo>

XinZhangBBL, 2025-04-23:
@Westoz123 Let me explain that the filament used in the model are ASA, which does not match the filament （PC in the video）in AMS, so mapping cannot be performed. If you want to use ASA to print your model, you need to install ASA in AMS.  If you want to use filament in AMS for printing, you can consider setting the consumables in the model as consumables in AMS.

XinZhangBBL, 2025-04-23:
@Westoz123 And you may also need to setup the filament type of the external spool . Since external spool does not support filament identification

XinZhangBBL, 2025-04-25:
Halo. The problem solved or not? 
The new version has been released, you can try new version. https://github.com/bambulab/BambuStudio/releases

Westoz123, 2025-04-25:
I have got it to work at times, I’m not sure how. Thanks.

XinZhangBBL, 2025-04-25:
@Westoz123  I wonder if these comments are helpful? 

[image]