bambulab/BambuStudio #6625: Sunlu profiles not available on H2D

Gluecksritter18, 2025-04-21:
### Bambu Studio Version

2.0.2

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I can't add/load the Sunlu Profiles inside Bambu studio for my H2D + AMS. It seems that the .json files for Sunlu must be updated.
I tried to copy them and add the H2D manually but that does not work sadly.

### Printer

Bambu H2D

### How to reproduce

Try to add Sunlu filament for a H2D printer.

### Actual results

Sunlu filament is not available for H2D

### Expected results

Sunlu filament profiles sould be available for H2D

### Project file & Debug log uploads

Not available 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Michael2808germ, 2025-05-04:
Same issue and with the generic profile I often get Spaghetti in the layers >4.
Hope this will get fixed soon as there is no prob with this filaments on my a1 mini and it´s not that high-priced printer model ;-)

sagiru, 2025-05-17:
Same here...i have a lot of Sunlu PLA spools with PLA Matte,  PLA Silk and PETG. Hopefully they will be available soon.

Regards

ImantsSkultans, 2025-06-17:
Any progress on this?

avboy72, 2025-07-19:
Same issue, H2D premium printer... fewer filament profiles than the basic ones?

elreydetodo, 2025-09-24:
Is it possible to manually import the profiles from the source? You can find them [here](https://github.com/bambulab/BambuStudio/tree/master/resources/profiles/BBL/filament/SUNLU). I'm going to manually create a profile based on the definition because I don't see any sort of import function.