bambulab/BambuStudio #6630: Single Object Multiple Files Alignment Issue

Chrisc2977, 2025-04-21:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro

### Additional system information

Not important as it is a software issue, present on multiple OS's and computers.
Same behaviour on mac and windows,

### Printer

All printers. Software bug.

### How to reproduce

Create a model and export seperate pieces for multiple colours (obj).

Drag files into Bambu Studio at the same time.

select appropriate filament colours.

Choose "yes" to import as a single object with multiple files.

Cry because they are not aligned properly :(

In previous versions of Bambu Studio the same file aligning properly.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

No log file needed.
No project file needed.

I have uploaded example file so you can test.
I have uploaded screenshots of the issue.

[Test Files.zip](https://github.com/user-attachments/files/19837196/Test.Files.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-22:
[image]
Hello, we have fixed it in the internal version. Thank you for your feedback. At present, you can import two objs using the v1.10 zip package version first, and then edit them with the latest version

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)