bambulab/BambuStudio #6634: Excessive Slicing Duration

seanbo, 2025-04-21:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU; Ryzen 7 9800
RAM: 64Gb DDR5
GPU: NVidia 4080ti

### Printer

Bambu A1 & A1 Mini

### How to reproduce

For problem:
1. Open Cleopatra3_Front_150x150.3mf with Bambu Studio
2. Click Slice
3. Wait for over an hour for it to finish.  The bulk of the time spent at "15% Wall generation"

To Confirm in Orca
1. Open Cleopatra3_Front_150x150.3mf with Orca Slicer
2. Hit Ok to error message about incompatible settings
3. Save as Cleopatra-Orca_Front_150x150.3mf
4. Click Slice
5. Slice completes in 4 minutes

To Double Check Bambu Studio
1. Open Cleopatra3-Orca_Front_150x150.3mf with Orca Slicer
2. Hit Ok to error message about incompatible settings
3. Save as Cleopatra-Orca=To-Bambu_Front_150x150.3mf
4. Click Slice
5. Slice completes in 8 minutes

All three versions included in below zip file

[Cleopatra.zip](https://github.com/user-attachments/files/19840433/Cleopatra.zip)

### Actual results

Results of process #1, over an hour to slice
Results of process #2, slices in 4 minutes in Orca Slicer
Results of process #3, using the file saved from Orca Slicer, slicing in Bambu Studio, then takes only 8 minutes (still double Orca Slicer, but at least a reasonable time).

### Expected results

Slice the file in less than 10 minutes, or relatively close to other comparable slicers...in this case Orca Slicer.

### Project file & Debug log uploads

[Cleopatra.zip](https://github.com/user-attachments/files/19840449/Cleopatra.zip)

[log.zip](https://github.com/user-attachments/files/19840453/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

XunZhangBambu, 2025-04-29:
@seanbo  We have found the problem and fix it internally. Currently you can disable "slow down for overhangs" to speed up the slice

[image]

seanbo, 2025-04-29:
> [@seanbo](https://github.com/seanbo) We have found the problem and fix it internally. Currently you can disable "slow down for overhangs" to speed up the slice
> 
> [image]

Thank you!

XunZhangBambu, 2025-06-03:
@seanbo  We have optimize the algorithm in our latest version 02.01.00.59

closed 2025-06-03 by XunZhangBambu