bambulab/BambuStudio #6642: 404 error

dariusserban, 2025-04-22:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

When i want to connect on my account, the tab is blank and nothing is running and also if i want to import a model from online models section it gives me a 404 message. 

The network ah been checks, i dont have any VPN active, the windows defender app has beed enabled and disabled, the Windows media player it installed.

everything i try it is not working... please let me know how to solve this problem. 

I wana let you all know that on the laptop, the windows has been restarted and reinstalled and after the reinstallation i'm haveing this problem, in the past the slicer worked fine and well. 

### Printer

Bambu Lab A!

### How to reproduce

1. Open the Slicer
2. Pred to connect button ( witch is not opening something)
3. try to search a model on online models section ( 404 error)

### Actual results

[image]
[image]

### Expected results

To solve somehow this problem

### Project file & Debug log uploads

[debug_network_Tue_Apr_22_10_20_13.log.zip](https://github.com/user-attachments/files/19845712/debug_network_Tue_Apr_22_10_20_13.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-22:
@dariusserban Sorry, could we try another network instead? If it still doesn't work, please let us know.

dariusserban, 2025-04-22:
@BoBoYu-BamBu i did, 

[image]

as you can see in this screen, I've connected the Laptop to the mobile hotspot in order to try again but the problem persists,

I've did as the wiki said to try to connect with the win files downloaded from the github, I've reached to connect the account, but the online models still don't work

dariusserban, 2025-05-04:
[image]

UPDATE!!!! For people who have this problem, 404 error on online models, go to preferences and set up the language to English