bambulab/BambuStudio #6643: 2 AMS with the same filament - printer print not with the choosen filament

ChristianBrand22, 2025-04-22:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I have two AMS and think found a little issue. I have in AMS1 (A) in slot 1 Bambulab white (, and in the AMS 2 (B) in slot 3 the same colour. 
When I chose a part and select the AMS 2 slot 3 whit colour, slice and start printing, then in the windows to start printing the filament A1 (AMS 1 slot 1) is shown. Even when I chose then B3 (AMS 2 slot 3) the printer chosse A1 to print ...?

Bambu Version: 2.0.1.50
Bambu Lab P1S: 01.07.00.00 (Newest)
AMS HUB: 00.00.00.43 (Newest)
AMS-1: 00.00.06.49 (Newest)
AMS-2: 00.00.06.49 (Newest)

[image]
[image]
[image]

### Printer

Bambulab P1S

### How to reproduce

Connect 2 AMS. 
Load The same filament in AMS1 slot 1 and in AM2 slot 3 and sync on Bambu studio.
Select a part and then select the filament 7 (= AMS 2 slot 3). 
Slice and print.

### Actual results

The printer print with AMS 1 slot 1 instead oof AMS 2 slot 3.

### Expected results

Print with the choosen filament.

### Project file & Debug log uploads

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-23:
Hello, although you selected B1 as the printing material in the preparation interface, due to the fact that A1 and B1 have exactly the same color, we have a concept of color distance in color matching. It happens that these two colors have exactly the same distance, and in this case, we first traverse to A1, so we matched A1
If you wish to use B1, you can manually select it. If A1 and B1 are both identical consumables, I think they can be printed with either one.

ThorsReign, 2025-04-24:
Hi @Haidiye00 of you check their final screenshot, they did select the B3 slot, the big they are reporting is that even though they selected the B3 slot to print from the printer chose to pull from A1 during printing instead of B3 so the actual print did not respect the selection.

fahaaaxiki, 2025-05-21:
Could you please zip up the entire log folder shown in the picture and provide it to us?

[image]

ChristianBrand22, 2025-05-21:
I think the upload here was ok, but I could not see the zip file....?

fahaaaxiki, 2025-05-23:
> I think the upload here was ok, but I could not see the zip file....?

Yes, we didn't see the log file on the ticket either. Could you please upload it again？ I can't reproduce this issue locally， so we have to analyze it through your log file.