bambulab/BambuStudio #6645: The program shuts down for no reason

dannyoakey, 2025-04-22:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 64 bit

### Additional system information

Intel(R) Core(TM) i7-3770K CPU @ 3.50 GHz 3.50 GHz
16 GB Ram
64 bit operative system.

### Printer

Bambu Lab A1 Mini

### How to reproduce

When I download and open a new print, it asks me to save the already opened project, when I choose an alternative a message comes up that it's an "Unhandled unknown exeption; terminating the application". I did not have this problem with the earlier version. This happens to me everytime I use the program.

### Actual results

[image]

### Expected results

It should be working and just open the new project.

### Project file & Debug log uploads

[crash_Tue_Apr_22_12_33_30_0_done.zip](https://github.com/user-attachments/files/19848267/crash_Tue_Apr_22_12_33_30_0_done.zip)

[001 Nintendo.zip](https://github.com/user-attachments/files/19848307/001.Nintendo.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

dannyoakey, 2025-04-22:
[001 Nintendo(2).zip](https://github.com/user-attachments/files/19848319/001.Nintendo.2.zip)

Haidiye00, 2025-04-22:
Thanks for your feedback. Can you still offer your dmp file and log file for us(? We will chceck if every time the crash happened at the same place.
And before we released the whole version, We checked 1.10 3mf import and no obvious crash happened so your problem is so wired.
Can you upload your all config file(like what in the red frame below). You can zip it and send to [malkey.chen@bambulab.com](mailto:malkey.chen@bambulab.com), we make sure no data leak and the only use is to find out this annoying problem.
[image]

dannyoakey, 2025-04-22:
[log.zip](https://github.com/user-attachments/files/19849689/log.zip)

Haidiye00, 2025-04-22:
[image]
Hello, we have identified the issue and the internal version has been fixed. The next version will bring it in~

dannyoakey, 2025-04-22:
Ok, thank you for the help :)

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

closed 2025-04-25 by Haidiye00