bambulab/BambuStudio #6647: H2D 0.2mm Nozzle currently uses 0.4mm Nozzle's layer height limits

illien3, 2025-04-22:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Printer

Bambu Lab H2D

### How to reproduce

Go to 'Prepare'
Go to 'Printer settings' with H2D active
Select 'Bambu Lab H2D 0.2 Nozzle'
Go to 'Left Extruder' or 'Right Extruder'
Under section 'Layer height limits'

### Actual results

0.2mm nozzle has a max layer height limits of 0.28mm, min of 0.08mm.
Which is the same as the 0.4mm nozzle
As a result, 0.06mm layer height presets are also missing for BBL H2D 0.2

### Expected results

As I understand, 0.2mm nozzle's min/max layer height should be 0.04 / 0.14 mm, which was the value used on BBL A1, P1, X1 series.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2025-05-13:
@illien3  thanks, we will correct the layer height limits in next version

XunZhangBambu, 2025-06-03:
@illien3 Hi, this has been fixed in our latest version 02.01.00.59